yorikvanhavre/FreeCAD-old #24: Selecting pathPocket on a circular pocket causes a segfault

dubstar-04, 2015-01-06:
Screenshot of the model. This is repeatable with a number of models, pocket diameters, depths and tools. 

If there is any more information I can provide please let me know. 

[image]

sliptonic, 2015-01-06:
confirmed.  
The debugger backtrace: 

#0  0x9b0687c6 in PathGui::ViewProviderPath::updateData(App::Property const*) () from /home/brad/FC/lib/PathGui.so
#1  0x9b06a000 in Gui::ViewProviderPythonFeatureT<PathGui::ViewProviderPath>::updateData(App::Property const*) () from /home/brad/FC/lib/PathGui.so
#2  0xb7acb066 in Gui::ViewProvider::update(App::Property const*) () from /home/brad/FC/lib/libFreeCADGui.so
#3  0xb7869d17 in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) () from /home/brad/FC/lib/libFreeCADGui.so
#4  0xb787c264 in boost::_mfi::mf2<void, Gui::Document, App::DocumentObject const&, App::Property const&>::operator()(Gui::Document*, App::DocumentObject const&, App::Property const&) const () from /home/brad/FC/lib/libFreeCADGui.so
#5  0xb787b39d in void boost::_bi::list3<boost::_bi::value<Gui::Document*>, boost::arg<1>, boost::arg<2> >::operator()<boost::_mfi::mf2<void, Gui::Document, App::DocumentObject const&, App::Property const&>, boost::_bi::list2<App::DocumentObject const&, App::Property const&> >(boost::_bi::type<void>, boost::_mfi::mf2<void, Gui::Document, App::DocumentObject const&, App::Property const&>&, boost::_bi::list2<App::DocumentObject const&, App::Property const&>&, int) ()
   from /home/brad/FC/lib/libFreeCADGui.so
#6  0xb787a2ca in void boost::_bi::bind_t<void, boost::_mfi::mf2<void, Gui::Document, App::DocumentObject const&, App::Property const&>, boost::_bi::list3<boost::_bi::value<Gui::Document*>, boost::arg<1>, boost::arg<2> > >::operator()<App::DocumentObject, App::Property>(App::DocumentObject const&, App::Property const&) ()
   from /home/brad/FC/lib/libFreeCADGui.so
#7  0xb78787f6 in boost::detail::function::void_function_obj_invoker2<boost::_bi::bind_t<void, boost::_mfi::mf2<void, Gui::Document, App::DocumentObject const&, App::Property const&>, boost::_bi::list3<boost::_bi::value<Gui::Document*>, boost::arg<1>, boost::arg<2> > >, void, App::DocumentObject const&, App::Property const&>::invoke(boost::detail::function::function_buffer&, App::DocumentObject const&, App::Property const&) () from /home/brad/FC/lib/libFreeCADGui.so
#8  0xb7347afd in boost::function2<void, App::DocumentObject const&, App::Property const&>::operator()(App::DocumentObject const&, App::Property const&) const ()
   from /home/brad/FC/lib/libFreeCADApp.so
#9  0xb7344e25 in boost::signals::detail::unusable boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >::operator()<boost::signals::detail::connection_slot_pair>(boost::signals::detail::connection_slot_pair const&) const () from /home/brad/FC/lib/libFreeCADApp.so
#10 0xb7342257 in boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator>::dereference() const ()
   from /home/brad/FC/lib/libFreeCADApp.so
#11 0xb733f89d in boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator>::reference boost::iterator_core_access::dereference<boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator> >(boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator> const&) () from /home/brad/FC/lib/libFreeCADApp.so
---Type <return> to continue, or q <return> to quit---
#12 0xb733c797 in boost::iterator_facade<boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator>, boost::signals::detail::unusable, boost::single_pass_traversal_tag, boost::signals::detail::unusable const&, int>::operator*() const () from /home/brad/FC/lib/libFreeCADApp.so
#13 0xb7338823 in boost::detail::postfix_increment_proxy<boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator> >::postfix_increment_proxy(boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals::detail::named_slot_map_iterator> const&) () from /home/brad/FC/lib/libFreeCADApp.so
#14 0xb733337f in boost::detail::postfix_increment_result<boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound2<void>::caller<App::DocumentObject const&, App::Property const&, boost::function<void (App::DocumentObject const&, App::Property const&)> >, b [... cut ...]
#82 0xb6007823 in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/i386-linux-gnu/libQtCore.so.4
---Type <return> to continue, or q <return> to quit---
#83 0xb6007b49 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/i386-linux-gnu/libQtCore.so.4
#84 0xb600d8fe in QCoreApplication::exec() () from /usr/lib/i386-linux-gnu/libQtCore.so.4
#85 0xb62f2a24 in QApplication::exec() () from /usr/lib/i386-linux-gnu/libQtGui.so.4
#86 0xb7847786 in Gui::Application::runApplication() () from /home/brad/FC/lib/libFreeCADGui.so
#87 0x0804f1d9 in main ()

yorikvanhavre, 2015-04-11:
Is this still crashing with recent code? If yes, could you provide a sample file and a step-by-step method to reproduce the crash?

sliptonic, 2015-04-11:
I don't have the segfault anymore but I have a new bug.  
Open the file linked below.
Switch to Path wb
Create a project
Select the floor of the circular pocket.
Try to create a Path Pocket.
For me, freecad locks up.  If I've launched from a terminal, I see it attempting to offset wires in what looks like an infinite loop.

https://dl.dropboxusercontent.com/u/9245675/circlepocket.fcstd

wood-galaxy, 2015-04-17:
I have the same behavior sliptonic describe.

yorikvanhavre, 2016-05-13:
Issue transferred to http://freecadweb.org/tracker/view.php?id=2550

closed 2016-05-13 by yorikvanhavre