yorikvanhavre/FreeCAD-old #29: Discretizing arcs

sliptonic, 2015-01-12:
Not all machine controllers can handle arcs natively.  So we should have a method to break arcs into discrete lines.  I'm not sure if this is the job of the post processor or if it should be done further upstream.

If it's to be one in the post, I think we should come up with a method and add it to the PathUtils.py file so it can be reused by the various posts.

If it's to be done upstream, then I assume the user has to indicate that they want output in that form and possibly a property to control the coarseness.

yorikvanhavre, 2015-01-24:
I think it should be done in post scripts, because It'll be much more efficient and interesting to keep storing these as arcs internally. Not only it takes much less space, but the discretization can be undone, or its resolution changed anytime...

Discretizing arcs is pretty easy, it's a matter of dividing the rotation between the start and end vertices by a certain amount of steps, and applying it incrementally to the start vector. That is what the ViewProviderPath does internally (since the opengl view cannot represent arcs).

We could do a generic function for that, that can be used in post scripts when needed...

sliptonic, 2015-01-24:
sounds good.  I think PostUtils.py should be the place for generic functions like that.

jonnor, 2015-09-06:
I did this in the rml_post (very quick and dirty), https://github.com/jonnor/FreeCAD/blob/rolandmodela/src/Mod/Path/PathScripts/rml_post.py#L123 function `xyarc`

yorikvanhavre, 2016-01-24:
Since the function now exists in @jonnor 's script, I think this issue can be closed... Reopen if needed!

closed 2016-01-24 by yorikvanhavre