yorikvanhavre/FreeCAD-old #30: Segfault after import

sliptonic, 2015-01-15:
Start FC clean so you're at the 'start' page.
Immediately open an .ngc file either by clicking it in the recent files list or by using the open file dialog.
Select pre processor and wait for the path to load.
Switch to Path WB. At this point, I get a segfault every time.
If I switch to path before importing, it works fine.

An Interesting side note:, if I open a file with the open dialog and select the preprocessor, it imports and zooms to show the full path.  

If I import the same file using the same post processor but initiate the import by clicking on it in the recent files list, it imports but doesn't zoom.

yorikvanhavre, 2015-01-24:
Problem 1) doesn't crash for me

Problem 2) also doesn't happen for me... Can you check that a " Gui.SendMsgToActiveView("ViewFit") " correctly appears in the python console after import? If not, it might be because the pre-processor failed

sliptonic, 2015-01-24:
Problem 1) does seem to be resolved.  I'm no longer crashing at that point.

problem 2) remains.  The viewfit message does not happen if I click from the 'recent files' list but does if I open it through the menu.  I'm using the example_pre script and in both cases it executes the 'done pre-processing' line.

yorikvanhavre, 2016-05-13:
Does problem 2) still happen to you? For me the opened file gets correctly zoomed, no matter if I open it from File->Open or from File->Recent

yorikvanhavre, 2016-05-13:
Issue transferred to http://freecadweb.org/tracker/view.php?id=2549

closed 2016-05-13 by yorikvanhavre