yorikvanhavre/FreeCAD-old #32: Feature: Stock Representation

sliptonic, 2015-01-20:
We would like to introduce the idea of a 'stock' object as part of the path project.

The stock represents that starting material that will be modified by the machine tool following the described paths.  

The stock object is useful for basic visualization since the user can see how feed and rapid moves intersect it.  It can also be queried by the path python features to set their properties.  For example, if the user is starting with a 12mm thick piece of material but reducing it to 10mm for the final part, an initial planing operation could set starting and ending depths by inspecting the stock and the part.

It is a simple object, either cuboid or cylindrical. Cuboid is sufficient for milling operations.  cylinders will be needed when we do turning operations.

The stock should, by default, displays with a high transparency so the original part solid(s) and path(s) can be seen through it.  

[image]

There should be only one stock object and it should only be an optional child of the PathProject:

[image]

Although there is only one stock object, clones should display at each placement specified by the fixture offsets.  If there are no fixture offsets, the stock object would be shown at 0,0,0

yorikvanhavre, 2015-01-24:
The stock could be an additional PropertyLink property to be added to the Project object, where you can add any shape-based object you want (ie. made with other FreeCAD workbenches)

sliptonic, 2015-02-10:
This has been implemented in a limited way.  Might need addtional work but I'm closing it for now.  Any additional features/bugs will be their own issue.

closed 2015-02-10 by sliptonic