yorikvanhavre/FreeCAD-old #34: RFC #1 - Multi-select and compounds

sliptonic, 2015-01-22:
This is neither a bug nor a feature.  It's nothing that needs to be worked on (yet).  It's an area that needs discussion and creative thinking so it's tagged RFC -- Request For Comment.

I've been thinking about the default behavior of the system when a user is creating a project involving lots of Path Python Features.  

Consider a test case:  The plate has eight holes, four of which are counter-sunk.
[image].  Assume that all the holes can be made with a circular pocket but that countersinking must be done first.

Setting up the project, the temptation will be for the user to:
* Select the edges of the four countersinks.
* Create a pocket path python feature.
* Select the eight edges of the smaller holes
* Create a second path python feature.

[image]

This results in two features in the tree, each with a path.  It works fine but when post processed will produce an inefficient path because the cutter has to go all the way around the piece twice. Lots of machine time is spent on unnecessary rapid moves.

The user recognizes this and wants to correct the problem.  To do so, he has to start over.  
* Select first countersink edge
* Create a pocket path python features
* Select two small holes
* Create a pocket path python features
* Select second countersink edge
* Create a pocket path python features
* Select two small holes
* Create a pocket path python features
...
Repeat for the other two group.

This results in eight path python features in the tree but produces an efficient path.  However, it's inefficient to create and maintain the project and there's lots of opportunity to screw up a setting.

[image]

A third possibility:
The user creates the project as in the first case above by selecting all the countersinks and creating one feature, then all the small holes and a second feature.  However, the path python feature recognizes the closed wires and creates compound containing individual paths for the four countersinks and a second compound with eight paths for the small holes.

Automatically created but non-optimized compounds:
[image]

When post processed, the same inefficient gcode is produced but the user can now manipulate the subpaths by rearranging them into new compounds to optimize machine travel.  This will require some help to make sure path properties like feed rate, step downs, etc are preserved but gives the user the ability to efficiently create project and generate efficient paths.

User can reorganize the sub-paths to optimize output. (The treeview doesn't show correct sorting but it is controlled in the project edit dialog)
[image]

yorikvanhavre, 2016-05-13:
Not sure we should transfer this to the tracker, not sure either if it is still valid given your recent work on Path @sliptonic . I'd like to transfer this to the official freecad world, though. Should I repost this on the forum?

sliptonic, 2016-05-14:
Let's not transfer it.  Things have changed a lot and much of this has been subsumed by new functionality.  Some of the issues raised still exist but will have to be attacked one at a time in different contexts.

closed 2016-05-14 by sliptonic