yorikvanhavre/FreeCAD-old #43: Hungarian translation problem only on Path workbench

krnkris, 2016-03-22:
I have problem, may be someone can help:

The Hungarian translation file seem to be corrupt,
only if I would make a Path.

If I open the Path workbench, then this .log file I get:

Log: Time = Sun Mar 20 14:02:49 2016
Log: AppDataSkipVendor = true
Log: AppHomePath = /usr/lib/freecad/
Log: AppIcon = freecad
Log: AppTempPath = /tmp/
Log: BinPath = /usr/lib/freecad/bin/
Log: BuildRepositoryURL = git://github.com/FreeCAD/FreeCAD.git master
Log: BuildRevision = 6635 (Git)
Log: BuildRevisionBranch = master
Log: BuildRevisionDate = 2016/03/19 14:55:09
Log: BuildRevisionHash = a4dc263846e2a06cf1a7bf79b51f37cc4add0b51
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 16
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
##### #### ### ####
# # # # # #
# ## #### #### # # # # #
#### # # # # # # # ##### # #
# # #### #### # # # # #
# # # # # # # # # ## ## ##
# # #### #### ### # # #### ## ## ##

Log: Debug = 0
Log: DocPath = /usr/lib/freecad/doc/
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.16
Log: LoggingFile = 1
Log: LoggingFileName = /home/hh/.FreeCAD/FreeCAD.log
Log: MaintainerUrl = http://www.freecadweb.org/wiki/index.ph ... =Main_Page
Log: PATH = /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
Log: PYTHONPATH =
Log: PythonSearchPath = :/usr/bin/../lib/python2.7/:/usr/bin/../lib/python2.7/plat-x86_64-linux-gnu:/usr/bin/../lib/python2.7/lib-tk:/usr/bin/../lib/python2.7/lib-old:/usr/bin/../lib/python2.7/lib-dynload
Log: RunMode = Gui
Log: SplashAlignment = Bottom|Left
Log: SplashInfoColor = #c8c8c8
Log: SplashScreen = freecadsplash
Log: SplashTextColor = #ffffff
Log: StartWorkbench = StartWorkbench
Log: SystemParameter = /home/hh/.FreeCAD/system.cfg
Log: UserAppData = /home/hh/.FreeCAD/
Log: UserHomePath = /home/hh
Log: UserParameter = /home/hh/.FreeCAD/user.cfg
Log: Verbose =
Log: Create Application
Log: Run App init script
Log: Init: starting App::FreeCADInit.py
Log: Init: Searching for modules...
Log: Init: Initializing /usr/lib/freecad/Mod/Sketcher... done
Log: Init: Initializing /usr/lib/freecad/Mod/ReverseEngineering... done
Log: Init: Initializing /usr/lib/freecad/Mod/Image... done
Log: Init: Initializing /usr/lib/freecad/Mod/PartDesign... done
Log: Init: Initializing /usr/lib/freecad/Mod/Ship(Init.py not found)... ignore
Log: Init: Initializing /usr/lib/freecad/Mod/Web... done
Log: Init: Initializing /usr/lib/freecad/Mod/MeshPart... done
Log: Init: Initializing /usr/lib/freecad/Mod/OpenSCAD... done
Log: Init: Initializing /usr/lib/freecad/Mod/Fem... done
Log: Init: Initializing /usr/lib/freecad/Mod/Spreadsheet... done
Log: Init: Initializing /usr/lib/freecad/Mod/Start... done
Log: Init: Initializing /usr/lib/freecad/Mod/Draft... done
Log: Init: Initializing /usr/lib/freecad/Mod/Test... done
Log: Init: Initializing /usr/lib/freecad/Mod/Import... done
Log: Init: Initializing /usr/lib/freecad/Mod/Complete... done
[... 141 lines cut ...]
'ascii' codec can't encode character u'\xe9' in position 9: ordinal not in range(128)

---------------------------------------

Can someone help me out to find , what causing the problem?

If I change the languages to a different one (english or German)
the then Path works.

Thank you!

OS: Ubuntu 14.04.4 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6635 (Git)
Build type: None
Branch: master
Hash: a4dc263846e2a06cf1a7bf79b51f37cc4add0b51
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17
------------------------------------------
Discussed on: 
http://forum.freecadweb.org/viewtopic.php?f=3&t=14914

krnkris, 2016-03-24:
Unfortunately problem not solved.

When I open a new Path, then it not opens in the right form on the Labels & attributes:

The Machine is NOT the children of Project, it is in the same row.

Then after it I can't add Tool.

It is again with the encoding:

R6651 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
##### #### ### ####
# # # # # #
# ## #### #### # # # # #
#### # # # # # # # ##### # #
# # #### #### # # # # #
# # # # # # # # # ## ## ##
# # #### #### ### # # #### ## ## ##

Log: Time = Thu Mar 24 09:49:40 2016
Log: AppDataSkipVendor = true
Log: AppHomePath = /usr/lib/freecad/
Log: AppIcon = freecad
Log: AppTempPath = /tmp/
Log: BinPath = /usr/lib/freecad/bin/
Log: BuildRepositoryURL = git://github.com/FreeCAD/FreeCAD.git master
Log: BuildRevision = 6651 (Git)
Log: BuildRevisionBranch = master
Log: BuildRevisionDate = 2016/03/22 18:25:38
Log: BuildRevisionHash = a69ee20f0c2acf146c935633ff43f933a8b14316
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 16
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
##### #### ### ####
# # # # # #
# ## #### #### # # # # #
#### # # # # # # # ##### # #
# # #### #### # # # # #
# # # # # # # # # ## ## ##
# # #### #### ### # # #### ## ## ##

Log: Debug = 0
Log: DocPath = /usr/lib/freecad/doc/
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.16
Log: LoggingFile = 1
Log: LoggingFileName = /home/hh/.FreeCAD/FreeCAD.log
Log: MaintainerUrl = http://www.freecadweb.org/wiki/index.ph ... =Main_Page
Log: PATH = /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
Log: PYTHONPATH =
Log: PythonSearchPath = :/usr/bin/../lib/python2.7/:/usr/bin/../lib/python2.7/plat-x86_64-linux-gnu:/usr/bin/../lib/python2.7/lib-tk:/usr/bin/../lib/python2.7/lib-old:/usr/bin/../lib/python2.7/lib-dynload
Log: RunMode = Gui
Log: SplashAlignment = Bottom|Left
Log: SplashInfoColor = #c8c8c8
Log: SplashScreen = freecadsplash
Log: SplashTextColor = #ffffff
Log: StartWorkbench = StartWorkbench
Log: SystemParameter = /home/hh/.FreeCAD/system.cfg
Log: UserAppData = /home/hh/.FreeCAD/
Log: UserHomePath = /home/hh
Log: UserParameter = /home/hh/.FreeCAD/user.cfg
Log: Verbose =
Log: Create Application
Log: Run App init script
Log: Init: starting App::FreeCADInit.py
Log: Init: Searching for modules...
Log: Init: Initializing /usr/lib/freecad/Mod/Sketcher... done
Log: Init: Initializing /usr/lib/freecad/Mod/ReverseEngineering... done
[... 164 lines cut ...]
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Log: [31muint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&)[0m: Condition failed: menu
Err: Running the Python command 'Path_Project' failed:
Traceback (most recent call last):
File "/usr/lib/freecad/Mod/Path/PathScripts/PathProject.py", line 136, in Activated
CommandProject.Create(incl)
File "/usr/lib/freecad/Mod/Path/PathScripts/PathProject.py", line 154, in Create
PathScripts.PathMachine.CommandPathMachine.Create()
File "/usr/lib/freecad/Mod/Path/PathScripts/PathMachine.py", line 219, in Create
_ViewProviderMachine(obj.ViewObject)
File "/usr/lib/freecad/Mod/Path/PathScripts/PathMachine.py", line 112, in __init__
vobj.addProperty("App::PropertyBool","ShowLimits","Path",translate("ShowMinMaxTravel","Switch the machine max and minimum travel bounding box on/off"))

'ascii' codec can't encode character u'\xe1' in position 3: ordinal not in range(128)Log: Active view is Nevtelen : 1[*] (at 0x2248aa0)
Err: *** Abort *** an exception was raised, but no catch was found.Err:
Err: ... The exception is:Err: SIGSEGV 'segmentation violation' detected. Address 8Err:

yorikvanhavre, 2016-05-13:
This is alreadty in the tracker: http://freecadweb.org/tracker/view.php?id=2524

closed 2016-05-13 by yorikvanhavre