yorikvanhavre/FreeCAD-old #47: ValueError while using importOBJ.insert()

greyltc, 2017-04-12:
FreeCAD 0.17, Libs: 0.17R10706 (Git)

Here's my script:
```python
#!/usr/bin/env python2
import sys
sys.path.append('/opt/freecad/lib')
import FreeCAD, importOBJ
mydoc = FreeCAD.newDocument("mydoc")
importOBJ.insert('/tmp/distributor8.obj','mydoc')
```

importOBJ.insert crashes with `ValueError: invalid literal for int() with base 10: '60//1'`
while reading line 2936 of my attached 'distributor8.obj' file. This file was exported from blender 2.78c

Can you help?

[distributor8.zip](https://github.com/yorikvanhavre/FreeCAD/files/916057/distributor8.zip)

greyltc, 2017-04-12:
Maybe this is the wrong place to put this bug report, I don't know.
Here it is again in possibly the right place: https://www.freecadweb.org/tracker/view.php?id=3008

yorikvanhavre, 2017-04-12:
Ok, I had never encountered this v/vt/vn notation before... https://en.wikipedia.org/wiki/Wavefront_.obj_file#Face_Elements
This problem is now fixed in b32e9806e
However your file contains non-planar polygons, which are skipped by the importer (I don't know a good algorithm to treat those). You can avoid that by checking the "triangulate all faces" option in Blender's OBJ exporter.

closed 2017-04-12 by yorikvanhavre